wenbowen123/iros20-6d-pose-tracking #1: When will the code be released

HalfSummer11, 2020-09-03:
Hi, thanks for the great work! I'm very interested in your project and I really look forward to the code release. Is there any plan on the release date? Thanks :))

wenbowen123, 2020-09-03:
@HalfSummer11 
Hi, thank you for your interest. I'm catching ICRA recently but I will try to release it by Oct.

HalfSummer11, 2020-09-04:
Thanks! By the way, I'm very impressed by the high frame rate of your model, especially when it has to render image on-the-fly (please correct me if I'm wrong). Is it mainly due to the small model w/ few parameters?

wenbowen123, 2020-09-04:
Generally, the framework runs fast because:
1) input size is small, only 176x176
2) network is indeed relatively shallow, and mostly 2D convolution.
3) rendering on-the-fly doesn't introduce much overhead since it's only needed in branch A, and is just basic OpenGL rendering (not photo-realistic ray tracing). Also the scene to render is very simple, just one object model without background.

HalfSummer11, 2020-09-05:
Got it :) Thank you!

closed 2020-09-05 by HalfSummer11

wenbowen123, 2020-10-19:
@HalfSummer11  the code has been released.

HalfSummer11, 2020-10-19:
Wow this is great! Thanks a lot for the work and for sharing the exciting news of code release 😊