wenbowen123/iros20-6d-pose-tracking #3: Setup requirements and instructions

xEnVrE, 2020-10-28:
Dear @wenbowen123,

first of all I would like to thank you for the great work!

I'm interested in trying the provided code and I was wondering whether you are planning to release more detailed instructions about installation requirements and steps to setup the environment.

Thank you

wenbowen123, 2020-10-30:
Hi @xEnVrE 

thank you for your interest. I was using an anaconda environment for managing python packages, if you want I can send you the environment, which is very big though (several GB). Or you can run the python file and install what's missing. I'm glad to help if you encounter any issue for setup.

xEnVrE, 2020-11-01:
Thank you for the answer. I'll be back in case of any issue during the setup.

closed 2020-11-01 by xEnVrE

greendddd, 2020-11-08:
> Hi @xEnVrE
>
> thank you for your interest. I was using an anaconda environment for managing python packages, if you want I can send you the environment, which is very big though (several GB). Or you [...]

Hi author,

Would you like to share the yaml file of your environment so that users could install the required packages? You could get the yaml file by running "conda env export > environment.yml" after you activated your env.

Thanks.

rammyaiyer05, 2020-11-12:
Hey @wenbowen123  amazing work!

I am trying to implement the code provided. It would be really great if you could provide the setup details as well as requirements details.

Thank you!

wenbowen123, 2020-11-12:
@xEnVrE @greendddd @rammyaiyer05 
Added the environment file. Please check the latest readme.

gchal, 2020-12-04:
> @xEnVrE @greendddd @rammyaiyer05
> Added the environment file. Please check the latest readme.

the environment.yml is failing

wenbowen123, 2020-12-08:
@gchal  Thanks for reporting this. There seems to be some [bug ](https://github.com/conda/conda/issues/9268) from anaconda. I will try to see if there is any better way to do this. Before that, you can try to install what's missing. The project should be compatible with most versions, and needs common libraries (opencv, pytorch, etc). The versions are listed in environment.yml. 

Sorry for the inconvenience.