wenbowen123/iros20-6d-pose-tracking #4: Multi-object tracking

pyni, 2020-11-05:
Nice work!
i have two questions:
1) i wonder how you do multi-object tracking shown in your video.
2) why Lie algebra is better than other alternatives to regress the rotation, such as quaternion or this paper[1]?

[1]Zhou, Y., Barnes, C., Lu, J., Yang, J., Li, H.: On the continuity of rotation representations in neural networks. In: Proceedings of the IEEE Conference on Computer Vision and Pattern Recognition. (2019) 5745–5753

wenbowen123, 2020-11-06:
1. For the comparison methods and se(3)-TrackNet, the tracking is applied to each object individually. The estimations are plotted together for visualization only. Sorry for the confusion. If there is a need for multi-object tracking, a naive way is to run multiple networks of se(3)-TrackNet simultaneously.
2. Good question. This representation has been found empirically numerical unstable for this task. On the other hand, Lie algebra representation is compact and easy to train to get satisfactory results. Thus it is used.

pyni, 2020-11-07:
Thanks for replying.

closed 2020-11-07 by pyni