wenbowen123/iros20-6d-pose-tracking #5: about  data

jinzhiyang1, 2020-12-02:
Hello, author. Thank you for sharing. I have some problems with datasets. You mentioned in the readme file that you need five datasets, but these add up to 350gb of memory, but my computer only has 200g. So I want to ask how to solve this problem?

jinzhiyang1, 2020-12-02:
Can you share the configuration path of ycbineoat dataset? If so. I will be very grateful

wenbowen123, 2020-12-03:
@jinzhiyang1 

1) you don't need to download all the data. You only need to download the data you need, depending on what you are hoping to do (training/testing/test on which object).

2) By "configuration path", are you saying the config.yml? Are you trying to train or evaluate on ycbineoat?

baojun701, 2020-12-03:
do you run successful? Can we have a wechat to communicate.18810443005

jinzhiyang1, 2020-12-03:
> @jinzhiyang1
>
>     1. you don't need to download all the data. You only need to download the data you need, depending on what you are hoping to do (training/testing/test on which object).
>
>     2. By [...]

Thank you for your reply. Yes, I want to train, test, and evaluate on ycbineoat. I have downloaded about 22 g, data sets of ycbineoat_ The organized dataset is about 15 g, but I don't know how to configure the path. Hope to get your reply again

wenbowen123, 2020-12-04:
@jinzhiyang1 

You can take a look at the predict.sh file.
```
CUDA_VISIBLE_DEVICES=0 \
python /media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/github/iros20-6d-pose-tracking/predict.py \
  --train_data_path /media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/github/end_to_end_tracking/Experiments/MINE/bleach_cleanser/train_data_blender_DR \
  --ckpt_dir /home/bowen/debug/bleach_cleanser/model_best_val.pth.tar \
  --mean_std_path /home/bowen/debug/bleach_cleanser \
  --class_id 12 \
```
The `train_data_path` should be pointed to the directory of [ycbineoat training data](https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCBInEOAT_traindata/) you downloaded (e.g. bleach_cleanser) 

`ckpt_dir` and `mean_std_path` are the set according to the downloaded pretrained weights from [ycbineoat](https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCBInEOAT_weights.zip). Since we are taking 'bleach_cleanser' as example, you should point to files inside the bleach_cleanser folder.

`class_id` is the object id in YCB dataset. Here bleach_cleanser is 12.

closed 2020-12-16 by wenbowen123