wenbowen123/iros20-6d-pose-tracking #9: about predict.py error

baojun701, 2020-12-06:
here is my path:
--YCBInEOAT_dir', default='/home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT/bleach0'
--train_data_path', default="/home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT_synthetic/bleach_cleanser/train_data_blender_DR
--ckpt_dir', default="/home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_weights/bleach_cleanser/model_best_val.pth.tar
--mean_std_path', default="/home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_weights/bleach_cleanser
I don't find bleach_cleanser folder in the YCBInEOAT , I think YCBInEOAT_dir stand for test_path,right or wrong?
I run the predict.py, but always account the eror " File "predict.py", line 218, in on_track
rgbB, depthB = crop_bbox(current_rgb, current_depth, bb, self.image_size)
File "/home/zkyd/iros20-6d-pose-tracking-master/Utils.py", line 340, in crop_bbox
color_crop[top_offset:bottom_offset, left_offset:right_offset, :] = color[top:bottom, left:right, :]
ValueError: could not broadcast input array from shape (71,633,3) into shape (71,0,3)
"
How can I solve it?
thank you if you can answer my question.

wenbowen123, 2020-12-06:
The paths look correct. Did you see this error at beginning or during tracking? Can you paste here the entire error?

baojun701, 2020-12-07:
(zky) zkyd@zkyd:~/iros20-6d-pose-tracking-master$ python3 predict.py 
ckpt_dir: /home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_weights/bleach_cleanser/model_best_val.pth.tar
dataset_info_path /home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT_synthetic/bleach_cleanser/train_data_blender_DR/../dataset_info.yml
/home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT/bleach0
self.object_cloud loaded and downsampled
self.object_width= 285.37860394994397
self.K: [[1.066778e+03 0.000000e+00 3.129869e+02]
 [0.000000e+00 1.067487e+03 2.413109e+02]
 [0.000000e+00 0.000000e+00 1.000000e+00]]
Loading ckpt from  /home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_weights/bleach_cleanser/model_best_val.pth.tar
pose track ckpt epoch=203
Using vispy renderer
model_path:  /home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT_synthetic/bleach_cleanser/textured.ply
self.cam_K:
 [[1.066778e+03 0.000000e+00 3.129869e+02]
 [0.000000e+00 1.067487e+03 2.413109e+02]
 [0.000000e+00 0.000000e+00 1.000000e+00]]
making dataset... for eval
#dataset: 0
self.trans_normalizer=0.03, self.rot_normalizer=0.5235987755982988
init pose
 [[-0.6981918   0.05113025  0.71419933 -0.39318141]
 [ 0.57201178  0.6396664   0.51329966  0.27422776]
 [-0.43062506  0.76665225 -0.47589299  0.83633948]
 [ 0.          0.          0.          1.        ]]
on_track:
 [[ 409 -371]
 [ 773 -371]
 [ 409   -7]
 [ 773   -7]]
Traceback (most recent call last):
  File "predict.py", line 643, in <module>
    predictSequenceMyData()
  File "predict.py", line 595, in predictSequenceMyData
    cur_pose = tracker.on_track(A_in_cam, rgb, depth, gt_A_in_cam=np.eye(4),gt_B_in_cam=np.eye(4), debug=debug,samples=samples)
  File "predict.py", line 218, in on_track
    rgbB, depthB = crop_bbox(current_rgb, current_depth, bb, self.image_size)
  File "/home/zkyd/iros20-6d-pose-tracking-master/Utils.py", line 340, in crop_bbox
    color_crop[top_offset:bottom_offset, left_offset:right_offset, :] = color[top:bottom, left:right, :]
ValueError: could not broadcast input array from shape (71,633,3) into shape (71,0,3)

> The paths look correct. Did you see this error at beginning or during tracking? Can you paste here the entire error?

wenbowen123, 2020-12-07:
It looks like your /home/zkyd/iros20-6d-pose-tracking-master/data/YCBInEOAT_Dataset/YCBInEOAT_synthetic/bleach_cleanser/train_data_blender_DR/../dataset_info.yml is actually from the ycb-video dataset, which is different from ycbineoat dataset. 

If you want to test on ycbineoat dataset, you need to download from [here](https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCBInEOAT_traindata/)

closed 2020-12-10 by wenbowen123