wenbowen123/iros20-6d-pose-tracking #10: about predict.sh

baojun701, 2020-12-08:
I update the path in the predict.sh, but it doesn't work.

ckpt_dir: None
dataset_info_path None/../dataset_info.yml
Traceback (most recent call last):
  File "predict.py", line 646, in <module>
    with open(dataset_info_path,'r') as ff:
FileNotFoundError: [Errno 2] No such file or directory: 'None/../dataset_info.yml'

here is my predict.sh:
CUDA_VISIBLE_DEVICES=1,2,3,4 \
python /home/zkyd/iros20-6d-pose-tracking-master/predict.py \
  --ttrain_data_path /home/zkyd/iros20-6d-pose-tracking-master/data/YCB_weights/banana/model_best_val.pth.tar \
  --mean_std_path /home/zkyd/iros20-6d-pose-tracking-master/data/YCB_weights/banana \
  --class_id 3 \

wenbowen123, 2020-12-10:
Did you pass "None" to ckpt_dir? You previous path configuration is correct. You only need to download the correct data. See https://github.com/wenbowen123/iros20-6d-pose-tracking/issues/9

baojun701, 2020-12-10:
I want to modify the path in the predict.sh to test other object, but it doesn't work. I modify the path in the predict.py(default), it's OK, I want to know why? the predict.sh is useless?

wenbowen123, 2020-12-10:
It's likely that you path is not set correctly [here](https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/0cdcc356786241fe746b29cba60c7cefe6abc0f1/predict.sh#L2), you need to change to your path to predict.py.

closed 2020-12-24 by wenbowen123