wenbowen123/iros20-6d-pose-tracking #11: ImportError: Library "GLU" not found

johnbhlm, 2020-12-10:
First of all, thank you very much for sharing the code.
I want to use your code to reproduce the test results in the paper, but I encountered some problems when configuring the environment:

> import pyrender
> Traceback (most recent call last):
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/__init__.py", line 334, in __getattr__
>     return getattr(self._module, name)
> AttributeError: 'NoneType' object has no attribute 'Window'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/__init__.py", line 334, in __getattr__
>     return getattr(self._module, name)
> AttributeError: 'NoneType' object has no attribute '_create_shadow_window'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/opt/conda/lib/python3.7/site-packages/pyrender/__init__.py", line 12, in <module>
>     from .viewer import Viewer
>   File "/opt/conda/lib/python3.7/site-packages/pyrender/viewer.py", line 36, in <module>
>     class Viewer(pyglet.window.Window):
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/__init__.py", line 340, in __getattr__
>     __import__(import_name)
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/window/__init__.py", line 1897, in <module>
>     gl._create_shadow_window()
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/__init__.py", line 340, in __getattr__
>     __import__(import_name)
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/gl/__init__.py", line 95, in <module>
>     from pyglet.gl.lib import GLException
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/gl/lib.py", line 149, in <module>
>     from pyglet.gl.lib_glx import link_GL, link_GLU, link_GLX
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/gl/lib_glx.py", line 46, in <module>
>     glu_lib = pyglet.lib.load_library('GLU')
>   File "/opt/conda/lib/python3.7/site-packages/pyglet/lib.py", line 164, in load_library
>     raise ImportError('Library "%s" not found.' % names[0])
> ImportError: Library "GLU" not found.

I referenced some methods  `apt-get install freeglut3-dev`  but it didn't work,

How can I solve it?
thank you if you can answer my question.

wenbowen123, 2020-12-10:
you need to install pyrender. You can refer to this [link](https://pyrender.readthedocs.io/en/latest/install/index.html)

johnbhlm, 2020-12-23:
thank you ,I have solved

closed 2020-12-23 by johnbhlm

assia855, 2023-08-07:
Hi @johnbhlm  how did you solve it please?