wenbowen123/iros20-6d-pose-tracking #12: error about train.py

johnbhlm, 2020-12-23:
`root@dev-john-yjaf-7595585b4-zqjtr:/data/new_build/code/Pose/se3_TrackNet/TrackNet# sh train.sh
output_path /data/new_build/code/Pose/se3_TrackNet/tmp/train_output
loaded dataset info from: /data/YCB/YCB_traindata/master_chef_can/train_data_blender_DR/../dataset_info.yml
self.cam_K:
 [[1.066778e+03 0.000000e+00 3.129869e+02]
 [0.000000e+00 1.067487e+03 2.413109e+02]
 [0.000000e+00 0.000000e+00 1.000000e+00]]
making dataset... for train
#dataset: 199587
self.trans_normalizer=0.03, self.rot_normalizer=0.08726646259971647
len(train_dataset)= 199587
Computing mean std for n=10000
<torch.utils.data.dataloader.DataLoader object at 0x7fb3d51caed0>
Traceback (most recent call last):
  File "train_HQ.py", line 109, in <module>
    for i, (data, target, A_in_cams, B_in_cams, rgbA, rgbB, maskA, maskB) in enumerate(train_loader):
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/opt/conda/lib/python3.7/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
IndexError: Caught IndexError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/data/new_build/code/Pose/se3_TrackNet/TrackNet/datasets.py", line 95, in __getitem__
    if rgbB.shape[0]!=self.dataset_info['resolution']:
IndexError: tuple index out of range
`
my config.yml
`data_augmentation:
  hsv_noise: [15,15,15]
  gaussian_noise:
    rgb: 2
    depth: 5
  gaussian_blur_kernel: 6
  depth_missing_percent: 0.4

learning_rate: 0.001
weight_decay: 0.000001
epochs: 300
loss_weights:
  trans: 1
  rot: 1

data_path: /data/YCB/YCB_traindata/master_chef_can/train_data_blender_DR
validation_path: /data/YCB/YCB_traindata/master_chef_can/validation_data_blender_DR
output_path: /data/new_build/code/Pose/se3_TrackNet/tmp/train_output
batch_size: 200
n_workers: 20
`

/data/YCB/YCB_traindata/master_chef_can/dataset_info.yml
`boundingbox: 10
camera:
  centerX: 312.9869
  centerY: 241.3109
  focalX: 1066.778
  focalY: 1067.487
  height: 480
  width: 640
distribution: gauss
models:
- 0: null
  model_path: /data/YCB/YCB_traindata/master_chef_can/textured.ply
object_width: 189.09586716088756
resolution: 176
train_samples: 200000
val_samples: 2000`

I check the data ,do not find error，Is there something unmodified that caused this error?

wenbowen123, 2020-12-23:
Can you check the shape of `rgbB` and where is the file it reads from? rgbB should be HxWx3.

johnbhlm, 2020-12-24:
I have solved it because of environmental problems.

closed 2020-12-24 by johnbhlm