wenbowen123/iros20-6d-pose-tracking #13: Conda Environment environment.yml issues

lachwot, 2021-01-01:
Hi!
Pardon if this is explained in detail somewhere else, however I keep running into conflicting packages when creating the conda environment using the environment.yml file. More specifically, Conda continuously attempts to solve the environment (for about half an hour at a time), giving me the following terminal output:

```
mu00185683@mu00185683-GL65-Leopard-10SFK:~/iros20-6d-pose-tracking$ conda env create -f environment.yml 
Collecting package metadata (repodata.json): done
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
```
It will then examine the conflicts for various packages present in the environment file and ultimately fail.

Using the default file straight from the repository, my terminal takes a while to attempt to solve the conflicts, where it will eventually end in a ResolvePackageNotFound error. It does not get any further than this and installing the listed packages manually does not work.

Hence, I have made changes to the file, which involve moving the following 3 packages:
`gcc_impl_linux-64=7.5.0=hd420e75_6`
`gxx_impl_linux-64=7.5.0=hdf63c60_6`
`qt=4.8.7=2`

from under the "dependencies" tag to under the "pip" tag to avoid these errors. This is as suggested by Aki1987 in this thread: https://github.com/datitran/object_detector_app/issues/41 . These three packages are the ones that appear in the ResolvePackageNotFound error.
This then brings me to this point where I am getting `Solving environment: failed" Errors and whatnot. The process takes about half an hour before it spits out a very long error (far longer than my terminals viewing window) meaning that I cannot figure out what is going on.

My setup:
Miniconda, installing the 64-bit Ubuntu 18.04 version following this link: https://docs.conda.io/en/latest/miniconda.html#linux-installers 
Python 3.8.5
pip 20.2.4 from /home/mu00185683/miniconda3/lib/python3.8/site-packages/pip (python 3.8)
I'm happy to provide more info if needed.

I'm very lost in this, and all I want to do is to be able to run the network and determine if it's the right fit for my application! If this environment setup is not needed, then please let me know because the instructions on actually running the network are very vague (But that's for another issue thread!)

I'll update this thread if I happen to figure it out.

Any help will be greatly appreciated!
Cheers,
Lachlan

wenbowen123, 2021-01-01:
@lachwot 
Hi, there is a known [issue ](https://github.com/conda/conda/issues/9268) of anaconda when installing bigger environment from yaml file. For this reason, to setup the environment, the recommended way would be directly run the python files and see if it's reporting any package missing error. Then install the missing package according to the version specified [here](https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/master/environment.yml). There aren't many packages actually dependent (much less than listed in the environment.yaml)

lachwot, 2021-01-01:
@wenbowen123  
Thanks for the response!
I see. 
Should I install the packages with pip3 or conda?

lachwot, 2021-01-02:
Good news, I've managed to make the python files run (at least until other, dataset related errors pop up)
Installing the packages manually with either pip3 and conda seemed to do the trick. 
Feel free to close if being able to run the python files without package errors was the intention behind the conda environment.

closed 2021-01-03 by wenbowen123