wenbowen123/iros20-6d-pose-tracking #16: about predict.py and evl_yab.py

zhuyazhi, 2021-01-09:
您好，请问这两个.py分别的目的是实现什么呢，有什么区别？
如果要进行测试，需要运行两个.py吗?

zhuyazhi, 2021-01-09:
我按照顺序先运行predict.py，但是出现以下error(我猜测是因为不能可视化)
Using vispy renderer
Traceback (most recent call last):
  File "predict.py", line 645, in <module>
    predictSequenceMyData()
  File "predict.py", line 566, in predictSequenceMyData
    tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir, trans_normalizer=0.03, rot_normalizer=30*np.pi/180)
  File "predict.py", line 163, in __init__
    self.renderer = VispyRenderer(dataset_info['models'][0]['model_path'], self.K, H=dataset_info['resolution'], W=dataset_info['resolution'])
  File "/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/vispy_renderer.py", line 50, in __init__
    app.Canvas.__init__(self, show=False, size=(W,H))
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/canvas.py", line 205, in __init__
    self.create_native()
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/canvas.py", line 222, in create_native
    self._app.backend_module.CanvasBackend(self, **self._backend_kwargs)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/backends/_pyglet.py", line 211, in __init__
    screen=self._vispy_screen)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/xlib/__init__.py", line 165, in __init__
    super(XlibWindow, self).__init__(*args, **kwargs)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/__init__.py", line 570, in __init__
    display = pyglet.canvas.get_display()
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/canvas/__init__.py", line 94, in get_display
    return Display()
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/canvas/xlib.py", line 123, in __init__
    raise NoSuchDisplayException('Cannot connect to "%s"' % name)
pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"
但是我注释了vispy_renderer.py中的几行之后这个错误没有了

出现了以下错误，您可以给我给一些指导吗。
[image]

zhuyazhi, 2021-01-09:
我按照顺序先运行predict.py，但是出现以下error(我猜测是因为不能可视化)
Using vispy renderer
Traceback (most recent call last):
File "predict.py", line 645, in
predictSequenceMyData()
File "predict.py", line 566, in predictSequenceMyData
tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir, trans_normalizer=0.03, rot_normalizer=30*np.pi/180)
File "predict.py", line 163, in init
self.renderer = VispyRenderer(dataset_info['models'][0]['model_path'], self.K, H=dataset_info['resolution'], W=dataset_info['resolution'])
File "/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/vispy_renderer.py", line 50, in init
app.Canvas.init(self, show=False, size=(W,H))
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/canvas.py", line 205, in init
self.create_native()
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/canvas.py", line 222, in create_native
self._app.backend_module.CanvasBackend(self, **self._backend_kwargs)
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/vispy/app/backends/_pyglet.py", line 211, in init
screen=self._vispy_screen)
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/xlib/init.py", line 165, in init
super(XlibWindow, self).init(*args, **kwargs)
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/init.py", line 570, in init
display = pyglet.canvas.get_display()
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/canvas/init.py", line 94, in get_display
return Display()
File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/canvas/xlib.py", line 123, in init
raise NoSuchDisplayException('Cannot connect to "%s"' % name)
pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"
但是我注释了vispy_renderer.py中的几行之后这个错误没有了

出现了以下错误，您可以给我给一些指导吗。
(se3) <Trial 1924760 worker_0> iros20-6d-pose-tracking $ python predict.py
ckpt_dir: /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/output/YCB_weights/bleach_cleanser/model_best_val.pth.tar
dataset_info_path /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_traindata/bleach_cleanser/train_data_blender_DR/../dataset_info.yml
/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_Video_Dataset/models/021_bleach_cleanser/textured.ply
[  29.86251    30.459349   28.622227 1591.2026    170.07364   160.50923
  146.59778   190.98444 ]
[  29.86251    30.459349   28.622227 1591.2026    170.07364   160.50923
  146.59778   190.98444 ]
/media/YCBInEOAT_Dataset/bleach0
self.object_cloud loaded and downsampled
self.object_width= 285.37860394994397
Loading ckpt from  /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/output/YCB_weights/bleach_cleanser/model_best_val.pth.tar
pose track ckpt epoch=112
Using vispy renderer
model_path:  /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_Video_Dataset/models/021_bleach_cleanser/textured.ply
self.cam_K:
 [[1.066778e+03 0.000000e+00 3.129869e+02]
 [0.000000e+00 1.067487e+03 2.413109e+02]
 [0.000000e+00 0.000000e+00 1.000000e+00]]
making dataset... for eval
#dataset: 0
self.trans_normalizer=0.03, self.rot_normalizer=0.5235987755982988
Traceback (most recent call last):
  File "predict.py", line 645, in <module>
    predictSequenceMyData()
  File "predict.py", line 573, in predictSequenceMyData
    pred_poses = [gt_poses[0]]
IndexError: list index out of range

zhuyazhi, 2021-01-09:
上面的问题是不是出在在两个数据集搞混了呢？
在predict.py中我该怎么选择在哪个数据集上测试

zhuyazhi, 2021-01-09:
我改成predictSequenceYcb()，也出现了类似的error:
(se3) <Trial 1924760 worker_0> iros20-6d-pose-tracking $ python predict.py
ckpt_dir: /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/output/YCB_weights/bleach_cleanser/model_best_val.pth.tar
dataset_info_path /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_traindata/bleach_cleanser/train_data_blender_DR/../dataset_info.yml
/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_Video_Dataset/models/021_bleach_cleanser/textured.ply
[  29.86251    30.459349   28.622227 1591.2026    170.07364   160.50923
  146.59778   190.98444 ]
[  29.86251    30.459349   28.622227 1591.2026    170.07364   160.50923
  146.59778   190.98444 ]
self.object_cloud loaded and downsampled
self.object_width= 285.37860394994397
Loading ckpt from  /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/output/YCB_weights/bleach_cleanser/model_best_val.pth.tar
pose track ckpt epoch=112
Using vispy renderer
model_path:  /opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_Video_Dataset/models/021_bleach_cleanser/textured.ply
self.cam_K:
 [[1.066778e+03 0.000000e+00 3.129869e+02]
 [0.000000e+00 1.067487e+03 2.413109e+02]
 [0.000000e+00 0.000000e+00 1.000000e+00]]
making dataset... for eval
#dataset: 0
self.trans_normalizer=0.03, self.rot_normalizer=0.08726646259971647
Traceback (most recent call last):
  File "predict.py", line 643, in <module>
    predictSequenceYcb()
  File "predict.py", line 455, in predictSequenceYcb
    print('gt_poses[0]=\n',gt_poses[start_frame])
IndexError: list index out of range

wenbowen123, 2021-01-10:
Hi, predict.py is used to predict pose results. eval_ycb.py will then evaluate its saved poses and get the scores on the benchmark.

zhuyazhi, 2021-01-11:
Thank you for your reply. 
1. I have gotten an adi_auc score after predict.py，is it a ADD score? So I can't clearly understand "get the scores on the benchmark", since the adi_auc is obtained by predict.py.
2. for each calss, if we want to use the pretrained weight to test on that class in ycb. for example(bleach_cleanser) , to test on it, I modify the seq_id = 51, class_id = 12. 51 and 12 is the number exist in the forlder "data_organized"。
so , actually we have no need to use the data(0000,...0091) in YCB_Video_dataset， but use the data in data_organized(0048,...0059) is it right?

def predictSequenceYcb():
	init = 'gt'
	seq_id = 51
	test_data_path = '{}/data_organized/%04d'.format(args.ycb_dir)%(seq_id)
	print(test_data_path)
	class_id = 12

zhuyazhi, 2021-01-11:
Run evl_ycb.py，I think seq_id can be directly = 51，and seq_id = int(pose_file.replace(args.res_dir,'').split('/') [0].replace('seq','')) got an error，I think this error is related to your path， so I let seq_id = 51 directly， and got a result 
>>>>>>>>>>>>>>>> args.class_id: 12 021_bleach_cleanser
add: 93.42925158905837
adi: 97.0202518718414
Is that right? expect your reply~

the code in evl_ycb.py I modified：
  for i in range(len(pose_files)):
    pose_file = pose_files[i]
    seq_id = 51   #int(pose_file.replace(args.res_dir,'').split('/')[0].replace('seq',''))
    frame_id = int(os.path.basename(pose_file).split('.')[0][:-2])+1
    print(frame_id)
    seq_frame_str = '%04d/%06d'%(seq_id,frame_id)
    if seq_frame_str not in keyframes:
      continue
    pred = np.loadtxt(pose_file)
    gt_file = '{}/data_organized/%04d/pose_gt/{}/%06d.txt'.format(args.ycb_dir,args.class_id)%(seq_id,frame_id)
    gt_pose = np.loadtxt(gt_file)

johnbhlm, 2021-01-14:
I think that different CAD models lead to different results.
In predict.py file, you may be using YCB_traindata->data_info.yml->textured.ply as CAD model，but in eval_ycb.py，you may be using *points.xyz as CAD model. So when calculating ADD and ADD-S will be different.

zhuyazhi, 2021-01-14:
thank you for your suggestion~

wenbowen123, 2021-01-14:
@johnbhlm  says is correct.

@zhuyazhi , evaluating only on keyframes is the rule of YCB-Video benchmark. You can check PoseCNN for more details on the evaluation protocol.

zhuyazhi, 2021-01-14:
ok, thanks for your reply。

closed 2021-01-14 by wenbowen123

jinzhiyang1, 2021-03-24:
> ok, thanks for your reply。
大佬你好，我最近也在看这个项目，但是在eval的时候出现了一些错误，你能分享一下eval_ycbineoat.py的配置路径么，万分感谢