wenbowen123/iros20-6d-pose-tracking #17: cannot find some functions and how to generate dataset_info.yml?

jingweirobot, 2021-01-14:
Dear Bowen
     We are trying to re-implement your work. But it is not available for the current repo.
     1.  The repo does not include completeBlenderYcbDR(), compute_2Dboundingbox, and normalize_scale functions. So the code cannot be ran successfully.
     2.  How to generate the dataset_info.yml files? The raw ycb-video dataset also does not have such files. 

Thanks in advance.

johnbhlm, 2021-01-14:
I find that:
  compute_2Dboundingbox() may be compute_bbox() in Utils.py file
  normalize_scale () may be crop_bbox() in Utils.py file

but I do not know what the values dataset_info['max_translation'] and dataset_info['max_translation'] in dataset_info.yml file

jingweirobot, 2021-01-14:
Maybe, 
I also do not know what's this.  We can wait for Wenbo's reply.

wenbowen123, 2021-01-14:
see https://github.com/wenbowen123/iros20-6d-pose-tracking/issues/15

jingweirobot, 2021-01-15:
@wenbowen123   @johnbhlm   

Thanks wenbo, ModelRendererOffscreen function cannot be found in the repo. Could you update this?

wenbowen123, 2021-01-16:
updated.

closed 2021-01-17 by jingweirobot