wenbowen123/iros20-6d-pose-tracking #18: No depth for training and inference

zhuyazhi, 2021-01-14:
sorry for bother you，in your paper "ablation study"，“For No depth, the depth modality is removed in both training
and inference stage to study its importance”。

how can I remove depth modality?

1. I found that in train.py:
	augmentations = Compose([
							HSVJitter(hsv_noise[0],hsv_noise[1],hsv_noise[2]),
							GaussianNoise(config['data_augmentation']['gaussian_noise']['rgb'], config['data_augmentation']['gaussian_noise']['depth']),
							GaussianBlur(config['data_augmentation']['gaussian_blur_kernel']),
							BlackCover(prob=0.2),
							# DepthMissing(prob=0.5,missing_percent=config['data_augmentation']['depth_missing_percent']),
							])
so I should remove the "#" and use DepthMissing(prob=0.5,missing_percent=config['data_augmentation']['depth_missing_percent']?

2. in predict.py, there are many places involves "depth", how should I modify the code to realize No depth for training and inference?
could you give me some advice?, thank you.

wenbowen123, 2021-01-14:
I would say the simplest way is the not send depth to the neural network. such that it only takes RGB 3 channels.

zhuyazhi, 2021-01-14:
thank you very much.
1. you mean i could change the input channel from 4 to 3, is my understand right?

2. But I find that there are many "depth" keywords in the code(such like depthA,depthB), would it lead to error if the generated data of mine don't have depth but only rgb?

I am appreciated if you could reply.

class Se3TrackNet(nn.Module):
	def __init__(self, image_size=174):
		super().__init__()
		self.rot_dim = 3

		self.convA1 = ConvBNReLU(C_in=4,C_out=64,kernel_size=7,stride=2)
		self.poolA1 = nn.MaxPool2d(kernel_size=3,stride=2,padding=1)
		self.convA2 = ResnetBasicBlock(64,64,bias=True)

		self.convB1 = ConvBNReLU(C_in=4,C_out=64,kernel_size=7,stride=2)
		self.poolB1 = nn.MaxPool2d(kernel_size=3,stride=2,padding=1)
		self.convB2 = ResnetBasicBlock(64,64,bias=True)
		self.convB3 = ResnetBasicBlock(64,64,bias=True)

wenbowen123, 2021-01-14:
Change to this
```
self.convA1 = ConvBNReLU(C_in=3,C_out=64,kernel_size=7,stride=2)
self.poolA1 = nn.MaxPool2d(kernel_size=3,stride=2,padding=1)
self.convA2 = ResnetBasicBlock(64,64,bias=True)

self.convB1 = ConvBNReLU(C_in=3,C_out=64,kernel_size=7,stride=2)
self.poolB1 = nn.MaxPool2d(kernel_size=3,stride=2,padding=1)
self.convB2 = ResnetBasicBlock(64,64,bias=True)
self.convB3 = ResnetBasicBlock(64,64,bias=True)
```

And make sure in the model forward, you are only sending first 3 channels of the tensor. I think this might be enough.

zhuyazhi, 2021-01-15:
ok, thanks for your careful reply.

wenbowen123, 2021-01-16:
1. it's data augmentation about depth. You can ignore it. 
3. You can provide a fake depth image with all zeros.

zhuyazhi, 2021-01-17:
ok, thanks.

closed 2021-01-17 by zhuyazhi