wenbowen123/iros20-6d-pose-tracking #20: about pyrenderer

zhuyazhi, 2021-01-17:
sorry to bother you.
1.  once I use pyrender, 
self.renderer = Renderer([obj_path],self.cam_K,dataset_info['camera']['height'],dataset_info['camera']['width'])

this error occurs:

Traceback (most recent call last):
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyrender/platforms/pyglet_platform.py", line 39, in init_context
    width=1, height=1)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/xlib/__init__.py", line 165, in __init__
    super(XlibWindow, self).__init__(*args, **kwargs)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/window/__init__.py", line 588, in __init__
    config = screen.get_best_config(config)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyglet/canvas/base.py", line 194, in get_best_config
    raise window.NoSuchConfigException()
pyglet.window.NoSuchConfigException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "produce_train_pair_data.py", line 269, in <module>
    completeBlenderYcbDR()
  File "produce_train_pair_data.py", line 220, in completeBlenderYcbDR
    producer = ProducerPurturb(dataset_info)
  File "produce_train_pair_data.py", line 86, in __init__
    self.renderer = Renderer([obj_path],self.cam_K,dataset_info['camera']['height'],dataset_info['camera']['width'])
  File "/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/offscreen_renderer.py", line 69, in __init__
    self.r = pyrender.OffscreenRenderer(self.W, self.H)
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyrender/offscreen.py", line 31, in __init__
    self._create()
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyrender/offscreen.py", line 149, in _create
    self._platform.init_context()
  File "/home/tiger/anaconda3/envs/se3/lib/python3.7/site-packages/pyrender/platforms/pyglet_platform.py", line 45, in init_context
    'internal error message was "{}"'.format(e)
ValueError: Failed to initialize Pyglet window with an OpenGL >= 3+ context. If you're logged in via SSH, ensure that you're running your script with vglrun (i.e. VirtualGL). The internal error message was ""

2. and I have some questions about :
    meta = np.load(rgb_file.replace('rgb.png','poses_in_world.npz'))
    class_ids = meta['class_ids']
    poses_in_world = meta['poses_in_world']
    blendercam_in_world = meta['blendercam_in_world']
    pos = np.where(class_ids==class_id)
    B_in_cam = np.linalg.inv(cvcam_in_blendercam).dot(np.linalg.inv(blendercam_in_world).dot(poses_in_world[pos,:,:].reshape(4,4)))

because my syn_data is generated by the code pvnet-rendering, so the pose for *B is save as a pkl(0.jpg,0.pkl)，so I directly extract the pose of .pkl as *B's pose(B_in_cam)，I don't have the transformation as your code(blendercam_in_world, poses_in_world....). what's more, I use vispyrender as render, and it won't get the error in question 1, but all of the *A is black, and it doesn't have object.

    data = pickle.load(open('/opt/tiger/zyz_6d_pose/iros20-6d-pose-tracking/media/YCB_traindata/ggl/blender_syn_sequence/mydataset_DR/{:07}RT.pkl'.format(i),'rb'))
    B_in_cam = data

so I don't know the error is related with vispyrender or my pose(extract the pose of .pkl as *B's pose(B_in_cam))?

could you give me some advice?

closed 2021-01-17 by zhuyazhi

gonalo99, 2022-05-12:
Hello! Do you mind sharing how you solved this problem?

week0425, 2022-06-09:
Hello, I also meet this error. Can you share the solution? Thank you!

gonalo99, 2022-06-09:
Hi! I solved this error by changing `-e "DISPLAY=unix:0.0"` to `-e "DISPLAY=$DISPLAY"` in the `run_container.sh` file inside the `docker` folder, you can try to see if it works for you.

assia855, 2023-08-07:
Hi @week0425 and @wenbowen123  I have the same error and the solution of @zhuyazhi didn't work for me. can you suggest to me another solution please.

wenbowen123, 2023-08-08:
@week0425 @assia855 are you running on a headless server?

assia855, 2023-08-08:
@wenbowen123 I'm running the docker