wenbowen123/iros20-6d-pose-tracking #21: about B_in_camera

zhuyazhi, 2021-01-18:
hello, I am sorry to bother you again, because I have a question that need you correction.

 my syn_data is generated by blender, and  the pose for *B is save as a pkl(e.g.,0.pkl)，so I directly extract the pose of .pkl as *B's pose(B_in_cam)，I don't have the transformation as your code(blendercam_in_world, poses_in_world....). 
do you think this is right?
meta = np.load(rgb_file.replace('rgb.png','poses_in_world.npz'))
class_ids = meta['class_ids']
poses_in_world = meta['poses_in_world']
blendercam_in_world = meta['blendercam_in_world']
pos = np.where(class_ids==class_id)
B_in_cam = np.linalg.inv(cvcam_in_blendercam).dot(np.linalg.inv(blendercam_in_world).dot(poses_in_world[pos,:,:].reshape(4,4)))
    data = pickle.load(open('/opt/tiger/zyz_6d_pose/iros20-6d-pose-

that's my modified code:
data = pickle.load(open('.../blender_syn_sequence/mydataset_DR/{:07}RT.pkl'.format(i),'rb'))
B_in_cam = data

could you give me some advice?

wenbowen123, 2021-01-18:
1. this looks correct to me, if your pkl saves the pose of B in camera (z axis forward, y axis down).
2. it looks like your generated A and B pair has too large motion. What's your setting about max_rotation?

zhuyazhi, 2021-01-19:
thank you~
1. yes, we use the opencv camera.

2. I set the same max_rotation as yours。strangely, i found that the pose I extract from my .pkl as B_in_cam is not the same as the B_in_cam saved in .npz after running produce_train_pair_data.py.
Does there exist some transformation before saving the pose B_in_cam I extract from my .pkl  into .npz?

wenbowen123, 2021-01-20:
The B_in_cam should be same. The original produce_train_pair_data.py may not change it. You can check where that change happens.

zhuyazhi, 2021-01-21:
ok, thank you, i will check it.
thanks for your kindly reply.

zhuyazhi, 2021-01-27:
hello, wenbo，I want to know if the speed 90 fps in the paper includes the time of rendering?

wenbowen123, 2021-01-28:
Yes, it includes all the steps.

zhuyazhi, 2021-01-29:
when doing inference， I think using the model textured_simple（1w+ faces）for rendering is suitable, is it right?
because the textured.ply includes 50w + faces, and it's time consuming when use this model to render at the last frame's pose, which can not achieve the speed reported in the paper。

wenbowen123, 2021-01-30:
Yes, that could affect the computation time. Besides, the image to render is small (176x176) and the whole process was optimized in C++. It also depends on what hardware you are using.

closed 2021-04-16 by wenbowen123