wenbowen123/iros20-6d-pose-tracking #22: Obtained results different from  Ours_YCB_results

ung264, 2021-03-04:
Thanks for sharing the great work! I'm trying to reproduce the result on YCB. I downloaded the pretrained model weights for power_drill and the synthetic training dataset. Use predict.py to get pose txt file for the sequence "data_organized/0054".  The visualization result looks OK. But when I checked the pose for a certain frame, the results are slightly different from results in Ours_YCB_results. I’ve tried both pyrenderer and vispy, both results are different from “Ours_YCB_results/Ours/035_power_drill/ycb_results_model_epoch215/seq54”. Do you have any clue where the difference come from? 

I also noticed that the default value of translation and rotation normalizer is 0.03 meter and 5 degree in predict.py. But in the paper you mentioned translation and rotation of 0.02m and 15 degree. I’ve tried to change the translation and rotation normalizer to 0.02m and 15 degree, but that results in a less stable result visually and the difference with your result is even larger. I guess the translation and rotation normalizer used in predict.py should be the same as the ones used while generating the synthetic dataset. Is it possible for you to provide the normalizer values used for generating your synthetic data?
[image]
[image]

Looking forward to your reply! Thanks in advance.

wenbowen123, 2021-03-05:
Thank you for your interest!
This value is a normalizer, which can be set arbitrarily as long as it's larger than the max possible motion and has to be consistent during train and test. On YCB, I remember we set to 0.03m and 5deg for training. So the second figure (vispy) corresponds to the configuration we used. Depending on the hardware and each run, the inference results could be slightly different, as what you observed, the 1e-3 magnitude is within the possible range though. Therefore, the "Ours_YCB_results" are provided for reference in case others want to make plots for comparison etc.

ung264, 2021-03-06:
Thank you for your prompt reply! Really appreciate everything you shared. It helps a lot!

closed 2021-03-06 by ung264