wenbowen123/iros20-6d-pose-tracking #24: about eval_ YCBInEOAT.py

jinzhiyang1, 2021-03-20:
Hello author, thank you for your sharing and help. I tried to train ycbineoat datasets in engineering and generate models, but when I ran eval_ YCBInEOAT.py
[image]
 When you want to evaluate the model score, you always report an error "track back (most recent call last):"

File "/home/mimashi1/iros20-6d-pose-tracking-master/eval_ ycbineoat.py ", line 126, in <module>

eval_ all(args)

File "/home/mimashi1/iros20-6d-pose-tracking-master/eval_ ycbineoat.py ", line 89, in eval_ all

assert len(pred_ files)==len(gt_ files),'#pred_ files:{}, #gt_ files:{}'.format(len(pred_ files),len(gt_ files))

UnboundLocalError: local variable 'gt_ files' referenced before assignment”

My path configuration is like this

“ parser.add_ argument('--YCBInEOAT_ dir', default='/home/mimashi1/iros20-6d-pose-tracking-master/data/bleach_ cleanser')

parser.add_ argument('--ycb_ dir', default='/media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/DATASET/Tracking/YCB_ Video_ Dataset')

parser.add_ argument('--class_ id',type=int,default=1)

parser.add_ argument('--res_ dir',type=str,default='/home/mimashi1/iros20-6d-pose-tracking-master/debug')”

I hope you can give me some suggestions and look forward to receiving your reply. Thank you.

jinzhiyang1, 2021-03-20:
Hello author, thank you for your sharing and help. I try to train ycbineoat dataset in engineering and generate model, but when I run eval_ YCBInEOAT.py  to evaluate model score, I always report error

My path configuration is like this

I hope you can give me some suggestions and look forward to receiving your reply. Thank you.

wenbowen123, 2021-03-23:
Hi, can you check if the "folders" is empty right after this? 
https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/39282d43aa06eeb6c7dca7677f6d1d73784ff610/eval_ycbineoat.py#L73

jinzhiyang1, 2021-03-23:
> Hi, can you check if the "folders" is empty right after this?
> 
> https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/39282d43aa06eeb6c7dca7677f6d1d73784ff610/eval_ycbineoat.py#L73

Thank you for your reply. This folder is not empty. Its contents are generated during the execution of predict.py
[image]
.

wenbowen123, 2021-03-23:
I see your res_dir's content is shown, but I didn't see "folders", can you print them? Basically, you need to make sure there are some folders (named with object's names) under the res_dir.

jinzhiyang1, 2021-03-23:
> I see your res_dir's content is shown, but I didn't see "folders", can you print them? Basically, you need to make sure there are some folders (named with object's names) under the res_dir.

Is that what it is?
[image]

jinzhiyang1, 2021-03-24:
I have read your reply carefully. Is it set like this? But the same mistake was reported.
[image]

wenbowen123, 2021-03-24:
Can you delete *.txt files under debug folder and see? You might have re-arranged the folder structure after you extract the files from the compression you downloaded.

jinzhiyang1, 2021-03-25:
Is that so, but there are still problems. I think it may be the problem of the path, but I'm not sure which files the path corresponds to. In addition, do I need to run other datasets? The content in bleach is generated in the prediction phase.
[image]

jinzhiyang1, 2021-03-25:
Thank you very much for your reply. When I learned about the code, I found that Eval was running_ ycbineoat.py When it comes to the YCB data set, please run eval_ ycbineoat.py Do you need to download the 265g YCB dataset?

wenbowen123, 2021-11-29:
No, you only need YCBInEOAT dataset and the CAD models from YCB-Video.

closed 2022-01-05 by wenbowen123