wenbowen123/iros20-6d-pose-tracking #27: about the network

zhuyazhi, 2021-05-15:
hello, sorry to bother you, there are two "self.convAB2 "in the code, I wonder if the second one is  "self.convAB3"? 
according to the network picture in the paper, there are two ResnetBasicBlocks， but in the forward(), only use self.convAB2() one time.
hoping you reply~
[image]
        self.convAB1 = ConvBNReLU(128, 256, kernel_size=3, stride=2) 
        self.convAB2 = ResnetBasicBlock(256, 256, bias=True)
        self.convAB2 = ResnetBasicBlock(256, 256, bias=True)

        ab = torch.cat((a, b), 1).contiguous() 
        ab = self.convAB1(ab)  
        ab = self.convAB2(ab)
        output['feature'] = ab

wenbowen123, 2021-06-22:
Thanks for pointing this out. This should be a typo. Only one self.convAB2 is valid.

zhuyazhi, 2021-06-26:
thanks for you reply~
I have another question to bother you, that is, a ycb object may have exist in several test_sets, such as "bleach cleanser'', its id is 12, and in data_organized/0051, 0054, 0055, 0057 all have "bleach cleanser''.
so my question is, the reported results(97.25) of "bleach cleanser'' is the average results in these 4 test_sets?
such as 
0051: 97
0054: 96.5
0055: 97.5
0058: 98
so the reported results of "bleach cleanser'' is (97+96+96.5+98)/4=97.25?
expect your reply~

wenbowen123, 2021-07-01:
Not exactly. The final result is the AUC by plotting all data in a curve. For more details on this, you can refer to PoseCNN. We followed the same evaluation.

wenbowen123, 2021-07-01:
Closing this issue for now. Feel free to open other issues if there is any other topic.

closed 2021-07-01 by wenbowen123