wenbowen123/iros20-6d-pose-tracking #32: Questions about running speed(90 Hz) and overfitting

cnJLiu, 2021-09-11:
Hi, thank you for your sharing and help! I have two questions and look forward to your reply.
1.	I want to know how your running speed reaches 90Hz and why when I run predict.py(use fuction getResultsYcb()) under RTX3090, the speed is much lower than 90Hz (the output time interval of every 100 frames is greater than 10 seconds)
[image]

2.	I want to know if you have completely completed 300 epochs training on the YCB-Video and Ycbineoat datasets, and whether 300 epochs may lead to overfitting(I think 300 epochs may be too much), For example, "ycb_results_model_epoch180" under the path "Ours/002_master_chef_can/ycb_results_model_epoch180" means that epoch180 is the best after training for 300 epochs?

wenbowen123, 2021-09-24:
1. It says "No OpenGL accelerate ....." from your output. There might be something wrongly configured.
2. Yes, early stopping can be used if overfitting is observed.

cnJLiu, 2021-09-26:
> 1. Hi Bowen, I am sorry to disturb you again, my running speed is still very slow after I install OpenGL accelerate,  as shown in the figure below, it took 484.85 seconds to track 1719 poses in the 58th YCB-Video, I want to know how to solve this problem, thank you very much and look forward to your reply.
[image]

wenbowen123, 2021-10-10:
Sorry for the late reply. 
1)Have you tried other GPUs, such as 2080 Ti or 1080 Ti? 3090 is relatively new and I'm not sure whether it has been well supported by many libraries.
2) Did you measure the start and end time of the entire evaluation? That'll include some IO time and others that's not quite related to the tracking itself.

wenbowen123, 2021-11-29:
replied on email

closed 2021-11-29 by wenbowen123

Liu-Xinhang, 2022-06-01:
> For better communication, I have sent you an email and look forward to your reply.

I have the same problem that it almost takes 0.1 second to render an image with OpenGL accelerate, so what's wrong with it?