wenbowen123/iros20-6d-pose-tracking #34: about first About the first frame of the annotation pose

jinzhiyang1, 2022-02-19:
Hello, relying on your sharing, I made a new prediction data containing rgb map, depth map, with camera internal reference. But it lacks the annotated pose of the first frame, I would like to ask how can I get the pose of the first frame (4*4 matrix)

wenbowen123, 2022-02-19:
Hi, you can use single image 6D pose estimation such as [this ](https://github.com/wenbowen123/icra20-hand-object-pose), by skipping the hand state estimation. Or you can also use some 6D pose annotation tool to get the pose, e.g. https://github.com/sakizuki/6DPoseAnnotator.

jinzhiyang1, 2022-02-21:
Thanks for your reply, but I have a few more questions
1. I want to get the file in bleach0/annotated_pose, like this；

[
[image]
](url)

How should I do that.

2. What does the information contained in this file mean, is it the 6d pose information of a single point? Or is it the 6d pose information of the whole model or something else?
I look forward to your reply.

wenbowen123, 2022-02-23:
1) These are annotated ground truth pose for evaluation. In our case, we labeled frame by frame. For you own test data, if you don't need to evaluate, you can simply ignore these.

2)6d pose information of the whole model. For YCBInEOAT dataset, the models are YCB objects.

jinzhiyang1, 2022-02-24:
Thank you for your reply, the above problem has been solved with your help.
But I have a new problem, how do I get the 6 parameters in "camera" in the dataset_info.yml file?

closed 2022-03-01 by wenbowen123