wenbowen123/iros20-6d-pose-tracking #36: Two questions about paper

elevenjiang1, 2022-03-17:
Hello bowen, I have two question about your paper:
1. How to make the YCBInEOAT dataset?
   In paper, you said that the dataset is "accurately annotated manually", can you provide detail tools or method for that?

2. Can one network track all objects?
   In 2D vision tracking like KCF or some deep learning method, the tracking pipeline is just draw a 2D bbox, then the network begin to tracking the 2D bbox. All kinds of objects can be used in a same network.
   Have you ever try to do so in 3D tracking? Maybe given a random object CAD and its inital pose, then tracking it in camera, but not just track one object base on one network parameters?

Thank you~
Best

wenbowen123, 2022-03-17:
Hi
1)We developed a semi-automatic annotation tool where user can provide 2D box as input and the 6D pose can be computed. We might release the code later. Stay tuned.

2)6D pose involves a larger and more complex solution space which makes it harder to generalize. But there is a trend in making it more generalizable. So I agree it'd be an interesting future work of this paper. For now, for real applications, if there is a need to track different objects, we just train multiple objects at the same time as each network is small, and that's how we did in [this paper](https://www.youtube.com/watch?v=3blr7u3VA7Y) and [this](https://www.youtube.com/watch?v=6QkRWfFt3RI).

elevenjiang1, 2022-03-17:
Thanks for your reply~

closed 2022-03-17 by elevenjiang1