wenbowen123/iros20-6d-pose-tracking #37: Blender file 1.blend does not exist

sunhaonuaa, 2022-03-21:
Hello bowen,
     I found that the blender file 1.blend does not exist in the repository when running your code. This may cause blender_main.py cannot be run correctly. Please check and upload the file.

Thank you~
Best

wenbowen123, 2022-03-21:
Thanks for reporting it! The readme has just been updated with the link to download.

sunhaonuaa, 2022-03-22:
Thanks for your reply~

closed 2022-03-22 by sunhaonuaa

ShougangShen, 2022-03-24:
Hello  bowen, I can't download the "Blender file 1.blend". The download link appears to be invalid.

Thank you!

ShougangShen, 2022-03-24:
Thanks for your reply.  
Sorry, I just used the link in Section ‘Data Download’ to download .
I found that the link in Section ‘Data Download’ is invalid, but the link in Section ‘Generate your own data’ can be downloaded.

wenbowen123, 2022-03-24:
I see, thanks for reporting!

elevenjiang1, 2022-04-20:
Sorry, but it seems that now 1.blend file can not download from Section 'Generate your own data'

wenbowen123, 2022-04-20:
@elevenjiang1  thanks for bringing it up! Yes our server breaks recently. It should come back later this week. Stay tuned!

wenbowen123, 2022-04-20:
@elevenjiang1 
the data should be back. Let me know if you still cannot access.

wenbowen123, 2022-04-20:
@elevenjiang1 
should be fixed. Let me know if you still cannot access.