wenbowen123/iros20-6d-pose-tracking #39: Results on YCBInEOAT dataset.

JiChun-Wang, 2022-03-27:
Hi, bowen, sorry to bother you again. When I run predit.py  to evaluate the nine video sequences of YCBInEOAT dataset. All sequences except for the 'bleach_hard_00_03_chaitanya' one perform well. I just use your provided pre-trained weights and the translation normalizer and translation normalizer are set to 0.03m and 30 degree respectively. The following are several visulization results:
[image]
[image]
![0000100](https:
[image]
//user-images.githubusercontent.com/41424692/160276945-597f4a23-9a82-4ee2-b8bf-7312f5176f3e.png)
[image]
Hope for your assistance.

wenbowen123, 2022-03-27:
This doesn't look like it's set up correctly. Can you make sure this is set to the correct `dataset_info.yml`?  It should be pointing to some file under the training data you downloaded from https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCBInEOAT_weights.zip
https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/1c28f02b93aa282a706181d1b28e9a56a9511d3c/predict.py#L630

wenbowen123, 2022-03-30:
I just verified from my side and didn't encounter this problem. From your image it looks like it lost tracking very early in the video. The scene so far in the video is not challenging, so it's still likely there is something wrong. Notice the `dataset_info.yml` here is different from the one used for YCB-Video data.

JiChun-Wang, 2022-04-08:
Thanks for your reply, I will try again.

closed 2022-04-08 by JiChun-Wang