wenbowen123/iros20-6d-pose-tracking #43: Arguments in predict_ros.py

eneserdo, 2022-05-25:
Could you please give more details about the usage of predict_ros.py?

Spesifically, what are these: 
- artifact_id
-  pose_init_file
- artifacts_folder
- camera_frame_name
- object_frame_name

wenbowen123, 2022-05-28:
Hi, that file is only used for ROS. `camera_frame_name` and `object_frame_name` are the ROS frame names. Otherwise if ROS isn't what you want, that file has similar usage as https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/master/predict.py
You can download the example data and run to get some idea about the rest of the parameters.

robertokcanale, 2022-06-24:
Hi, have you actually run it with ROS? it does look like ROS2, correct?

Did you just copy the file to your own ROS repo or did you just add the path to your workspace? I'm trying to understand the line:
`sys.path.append('/root/catkin_ws/src')`

Could you please provide me with more info on this? Also, which topics does it publish? Where is it possible to see the bounding boxes?
Could you provide a slightly more extensive guide on this? I am working on it right now a a bit more guidance would be appreciated, many thanks!
(Reading someonelse's code is always a bit tricky :)    )

wenbowen123, 2022-07-01:
This is ROS1.
You need to run this repo inside our provided docker.
The topic names and the tf names are defined around here https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/2df96b720e8e499b9f0d5fcebfbae2bcfa51ab19/predict_ros.py#L76

closed 2022-07-09 by wenbowen123