wenbowen123/iros20-6d-pose-tracking #46: questions regard setup

Liesy, 2022-08-16:
Hi, thanks for your public code. Errors occurred when launching docker container:

> Error: No such container: se3_tracknet
> access control disabled, clients can connect from any host
> run_container.sh: line 3: nvidia-docker: command not found

What's more, I'm trying to run your code in remote linux server that has no screen to display rendered results. So could you tell me how to run your code `predict.py/predictSequenceYcb()` without the assistance of docker?

Liesy, 2022-08-16:
Sry to bother you, I have solved problems by myself and your code can run in remote linux server with the help of vcxsrv and egl.

closed 2022-08-16 by Liesy

ZisongXu, 2024-02-19:
Sorry can i ask how did you solve this problem?