wenbowen123/iros20-6d-pose-tracking #51: TypeError: float() argument must be a string or a number, not 'open3d.open3d.geometry.PointCloud'

Chunsheng13, 2023-01-06:
when I run predict.py and predict_ros.py it comes the same error:

root@xiufeij-System-Product-Name:/home/se3_tracknet# python predict.py 
ckpt_dir: weights/YCB_weights/bleach_cleanser/model_best_val.pth.tar
dataset_info_path ./dataset_info.yml
YCBInEOAT/bleach0
self.object_cloud loaded and downsampled
Traceback (most recent call last):
  File "predict.py", line 663, in <module>
    predictSequenceMyData()
  File "predict.py", line 566, in predictSequenceMyData
    tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir, trans_normalizer=0.03, rot_normalizer=30*np.pi/180)
  File "predict.py", line 136, in __init__
    object_max_width = compute_obj_max_width(self.object_cloud)
  File "/home/se3_tracknet/Utils.py", line 450, in compute_obj_max_width
    return compute_cloud_diameter(model_cloud) * 1000
  File "/home/se3_tracknet/Utils.py", line 102, in compute_cloud_diameter
    hull = ConvexHull(points)
  File "qhull.pyx", line 2421, in scipy.spatial.qhull.ConvexHull.__init__
  File "/usr/local/lib/python3.6/dist-packages/numpy/core/_asarray.py", line 177, in ascontiguousarray
    return array(a, dtype, copy=False, order='C', ndmin=1)
TypeError: float() argument must be a string or a number, not 'open3d.open3d.geometry.PointCloud'

is it numpy version problem or something else?
hope you can me figure out what happent

wenbowen123, 2023-01-06:
I fixed the bug. Can you please try again?

Chunsheng13, 2023-01-07:
Thank you very much for your helping, should I re download the docker image or something else?

wenbowen123, 2023-01-07:
No need. You can directly run.

closed 2023-06-01 by wenbowen123