wenbowen123/iros20-6d-pose-tracking #55: Epoch is not saved to the Torch checkpoint

martinlyra, 2023-05-17:
Whenever `predict_ros.py` is run with our own trained data:
```
Loading ckpt from  /home/marcusmartin/artifacts/0/model_best_val.pth.tar
Traceback (most recent call last):
  File "predict_ros.py", line 109, in <module>
    tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir,trans_normalizer=dataset_info['max_translation'],rot_normalizer=dataset_info['max_rotation'])
  File "/home/marcusmartin/iros20-tracking/predict.py", line 152, in __init__
    print('pose track ckpt epoch={}'.format(checkpoint['epoch']))
KeyError: 'epoch'
```

The culprit seems to be
https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/1b63d0d16b771c4654bf9e571ede50baf7eda7a1/problems.py#L144
and
https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/1b63d0d16b771c4654bf9e571ede50baf7eda7a1/problems.py#L151

Our torch version when we trained the data was `2.0.1` - if you were using an older version, may I suggest you to add a [requirements.txt](https://pip.pypa.io/en/stable/reference/requirements-file-format/) too?

wenbowen123, 2023-05-18:
can you pull the code and retry? This issue should not be related to pytorch version. Are you running inside the docker?

martinlyra, 2023-05-19:
Nope, we run natively on the robot. - The fix seems to work fine!

closed 2023-05-19 by martinlyra