wenbowen123/iros20-6d-pose-tracking #59: docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]]. ERRO[0000] error waiting for container:

monajalal, 2023-05-30:
```
(base) mona@ard-gpu-01:~$ sudo  docker pull wenbowen123/se3_tracknet:latest
[sudo] password for mona: 
latest: Pulling from wenbowen123/se3_tracknet
976a760c94fc: Pull complete 
c58992f3c37b: Pull complete 
0ca0e5e7f12e: Pull complete 
f2a274cc00ca: Pull complete 
708a53113e13: Pull complete 
6295ebf8c067: Pull complete 
378a3c4b6a48: Pull complete 
39653001dda5: Pull complete 
1b7b24954ce9: Pull complete 
cde12d6e6d0c: Pull complete 
f9443eb23477: Pull complete 
5e072bee3770: Pull complete 
2079e9b75e82: Pull complete 
dc290c03a08b: Pull complete 
19ec1e4b0faa: Pull complete 
7575467a4504: Pull complete 
0b6f3b738ece: Pull complete 
bcbba2fb1faf: Pull complete 
b879d1017abb: Pull complete 
397c2a08ecfb: Pull complete 
c75519554b0d: Pull complete 
6259b7658b36: Pull complete 
3084d2a13df3: Pull complete 
f079557e4ff5: Pull complete 
8f44a8cc53a4: Pull complete 
Digest: sha256:cca11d217355b5ecc67d30c25f813a94dc7c3b5e57084542af4ef7f91106608f
Status: Downloaded newer image for wenbowen123/se3_tracknet:latest
docker.io/wenbowen123/se3_tracknet:latest

(base) mona@ard-gpu-01:~$ sudo docker run -it wenbowen123/se3_tracknet:latest
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/27c8edc2670632acf9df8369ff3ddcac81ce60339a73761f91ced6c4d64f2215/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: <nil>: unknown.
ERRO[0000] error waiting for container:            

(base) mona@ard-gpu-01:~$ sudo docker run --gpus all  wenbowen123/se3_tracknet:latest
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
ERRO[0000] error waiting for container:     
```

I have the following in `daemon.json`:
```
(base) mona@ard-gpu-01:~$ cat /etc/docker/daemon.json 
{
    "insecure-registries":["ardregistry:5000"],
    "experimental" : true,
    "default-runtime": "nvidia",
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}

(base) mona@ard-gpu-01:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Jun__8_16:49:14_PDT_2022
Cuda compilation tools, release 11.7, V11.7.99
Build cuda_11.7.r11.7/compiler.31442593_0
(base) mona@ard-gpu-01:~$ docker --version
Docker version 24.0.2, build cb74dfc

```

wenbowen123, 2023-05-30:
can you try the `run_container.sh` as mentioned [here](https://github.com/wenbowen123/iros20-6d-pose-tracking/tree/master#quick-setup)?

monajalal, 2023-05-30:
`(base) mona@ard-gpu-01:~/iros20-6d-pose-tracking/docker$ sudo bash run_container.sh`
worked
thank you

closed 2023-05-30 by monajalal