wenbowen123/iros20-6d-pose-tracking #62: error in running  python blender_main.py  -- Color management: image colorspace "sRGB OETF" not found, will use default instead.

monajalal, 2023-06-05:
I have no problem opening or using Blender. However, I get this error:
```
(base) mona@ard-gpu-01:~/iros20-6d-pose-tracking$ python blender_main.py 
/usr/bin/blender /home/mona/iros20-6d-pose-tracking/1.blend -b -P /home/mona/iros20-6d-pose-tracking/blender_dataset_generator.py 
Color management: using fallback mode for management
Color management: Error could not find role data role.
Blender 3.0.1
Read prefs: /home/mona/.config/blender/3.0/config/userpref.blend
Color management: scene view "Filmic" not found, setting default "Standard".
Read blend: /home/mona/iros20-6d-pose-tracking/1.blend
Color management: scene view "sRGB OETF" not found, setting default "Standard".
Color management: sequencer colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
SyntaxError: encoding problem: future_fstrings

Blender quit
```

monajalal, 2023-06-05:
remove the first line here
```
(tracknet) mona@ard-gpu-01:~/iros20-6d-pose-tracking$ vi blender_dataset_generator.py 
```

closed 2023-06-05 by monajalal

Wahaha-code, 2024-01-18:
delete ‘import bpy’？