wenbowen123/iros20-6d-pose-tracking #63: Failed to initialize Pyglet window with an OpenGL >= 3+ context.

assia855, 2023-08-08:
Hi @wenbowen123 thanks for sharing this awesome project, I'm struggling with this error:

`Using pyrenderer
model_path benchmarks/YCBInEOAT/bleach_cleanser/textured_simple.obj
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyrender/platforms/pyglet_platform.py", line 39, in init_context
    width=1, height=1)
  File "/usr/local/lib/python3.6/dist-packages/pyglet/window/xlib/__init__.py", line 171, in __init__
    super(XlibWindow, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pyglet/window/__init__.py", line 591, in __init__
    display = pyglet.canvas.get_display()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/canvas/__init__.py", line 94, in get_display
    return Display()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/canvas/xlib.py", line 123, in __init__
    raise NoSuchDisplayException('Cannot connect to "%s"' % name)
pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "iros20-6d-pose-tracking/predict.py", line 641, in <module>
    predictSequenceMyData()
  File "iros20-6d-pose-tracking/predict.py", line 567, in predictSequenceMyData
    tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir, trans_normalizer=0.03, rot_normalizer=30*np.pi/180)
  File "iros20-6d-pose-tracking/predict.py", line 161, in __init__
    self.renderer = Renderer([dataset_info['models'][0]['obj_path']],self.K,cam_cfg['height'],cam_cfg['width'])
  File "/home/iros20-6d-pose-tracking/offscreen_renderer.py", line 69, in __init__
    self.r = pyrender.OffscreenRenderer(self.W, self.H)
  File "/usr/local/lib/python3.6/dist-packages/pyrender/offscreen.py", line 31, in __init__
    self._create()
  File "/usr/local/lib/python3.6/dist-packages/pyrender/offscreen.py", line 149, in _create
    self._platform.init_context()
  File "/usr/local/lib/python3.6/dist-packages/pyrender/platforms/pyglet_platform.py", line 45, in init_context
    'internal error message was "{}"'.format(e)
ValueError: Failed to initialize Pyglet window with an OpenGL >= 3+ context. If you're logged in via SSH, ensure that you're running your script with vglrun (i.e. VirtualGL). The internal error message was "Cannot connect to "None""
`
How can I solve this error please.
Thanks in advance.

wenbowen123, 2023-08-08:
Are you running on a server without display?

assia855, 2023-08-09:
I added this line at the beginning of the predict.py script: `os.environ['PYOPENGL_PLATFORM'] = 'egl'`  
and now I have this error: 
`root@9f509805cf64:/home# ./iros20-6d-pose-tracking/predict.sh 
ckpt_dir: benchmarks/YCBInEOAT/YCBInEOAT_weights/bleach_cleanser/model_best_val.pth.tar
dataset_info_path benchmarks/YCBInEOAT/bleach_cleanser/train_data_blender_DR/../dataset_info.yml
/media/bowen/e25c9489-2f57-42dd-b076-021c59369fec/catkin_ws/src/iros20_dataset/video_rosbag/IROS_SELECTED/FINISHED_LABEL.iros_submission_version/bleach0
self.object_cloud loaded and downsampled
self.object_width= 285.37860394994397
Loading ckpt from  benchmarks/YCBInEOAT/YCBInEOAT_weights/bleach_cleanser/model_best_val.pth.tar
pose track ckpt epoch=203
Using pyrenderer
model_path benchmarks/YCBInEOAT/bleach_cleanser/textured_simple.obj
self.cam_K:
 [[319.58200073   0.         320.21498477]
 [  0.         417.11868286 244.34866809]
 [  0.           0.           1.        ]]
making dataset... for eval
#dataset: 0
self.trans_normalizer=0.03, self.rot_normalizer=0.5235987755982988
Traceback (most recent call last):
  File "iros20-6d-pose-tracking/predict.py", line 641, in <module>
    predictSequenceMyData()
  File "iros20-6d-pose-tracking/predict.py", line 574, in predictSequenceMyData
    pred_poses = [gt_poses[0]]
IndexError: list index out of range
`
I don't now if it relate to the line that I added or something else, thanks in advance for any help.

wenbowen123, 2023-08-09:
can you check if YCBVideo sequence works for you? https://github.com/wenbowen123/iros20-6d-pose-tracking/issues/53

assia855, 2023-08-11:
Hi @wenbowen123 thanks for your replay, I tried with  YCBVideo sequence but I still struggle with the same error. Any other suggestions please. 
Thanks in advance.

wenbowen123, 2023-08-12:
From the error it seems like the init pose was not successfully loaded. Can you check if this is empty? https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/9bc636808aa4091fa7b3ffed43215c349c49258c/predict.py#L446C12-L446C12

assia855, 2023-08-16:
Hi @wenbowen123 tank you so much for your help. I solved the problem of loading the position but the function of Tracker can't seem to function because the display couldn't connect with my laptop as I could understood. So I'm still stuck with this error: 
[image]
Thanks in advance for your big help.