wenbowen123/iros20-6d-pose-tracking #70: IndexError: list index out of range

aThinkingNeal, 2024-01-19:
Hi I am running the docker image but got this IndexError, could you take a look at it?

Thank in advance for your help!

[image]

the content of test.sh is:

[image]

wenbowen123, 2024-01-19:
[image]
from your screenshot, the dataset# is 0. This looks suspicious, can you confirm the data dir is given correctly?

aThinkingNeal, 2024-01-22:
Thanks for your help!

Usually what should the #dataset look like?

If data dir means the --train_data_path parameter:

I have run the command "ls" on the --train_data_path parameter path, the result looks fine to me. So I guess the data dir is given correctly?

[image]

My process of getting this data is as follows:

1. download the dataset of bleach_cleanser from this website:
https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCB_traindata/

2. extract it under the YCBV_data folder

---
The following point may or may not be relevant to this issue:

in the repo there is the data_organized dataset, but I have no idea on how to use it.

wenbowen123, 2024-01-23:
https://archive.cs.rutgers.edu/archive/a/2020/pracsys/Bowen/iros2020/YCB_traindata/
this is the training data. If you want to run inference, you need to download the `data_organized` which is an organized version of YCB-Video dataset. If you have configured those correctly, expect to see `#dataset` >0

aThinkingNeal, 2024-01-25:
[image]

this is my current dataset configuration, I have downloaded the `data_organized` and put them under the `ycb_dir` folder, which in my case is `/home/neal/debug/YCBV_data`.
(By the way, Could you share your dataset folder structure? It seems a bit unclear from the README session, I am afraid that others may face the same issue when setting up their dataset configuration.)

However, I am still getting the error: `IndexError: list index out of range`

[image]

I have tried to debug by adding a print statement in the predict.py:

[image]

it seems that the program didn't manage to go to the `test_data_dir` step, otherwise the print should show up in the error message.

Below is my command to run the program, note that I have updated the `ycb_dir`.

[image]

wenbowen123, 2024-01-26:
can you print `args.ycb_dir` here?
https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/18dc5bac30024e4d44465f3202ba6ffbb62e8e3b/predict.py#L315

If it's not updated like what you passed in the command line, maybe it's argparse's issue. Then try to directly edit it [here](https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/18dc5bac30024e4d44465f3202ba6ffbb62e8e3b/predict.py#L631)

aThinkingNeal, 2024-01-26:
Thanks for the help, I have managed to solve the `IndexError` problem. 
Essentially it is the `test_data_path` is not correctly configured.

Here is what I changed to make it work:

## First step: 
change the line of code from [here](https://github.com/wenbowen123/iros20-6d-pose-tracking/blob/18dc5bac30024e4d44465f3202ba6ffbb62e8e3b/predict.py#L450)

to 

`test_data_path = '{}/data_organized/%04d'.format(args.ycb_dir)%(seq_id)`

## Second Step:

organized my dataset configuration as:

`/home/neal/debug/YCBV_data/data_organized/0051`

In this way the test dataset is correctly passed and the problem is solved

I think the confusion comes from the README and codebase mismatch:

Here in the README session implies that should be a `data_organized` folder

[image]

However in the codebase this folder path is not reflected.

Hope this helps!

--- 

By the way, after solving this issue I encountered another issue, but I will continue in this place. #49

closed 2024-01-26 by aThinkingNeal