wenbowen123/iros20-6d-pose-tracking #71: predict_ros.py error finding model path

ifeeney, 2024-01-20:
Hi,

I was trying to run predict_ros.py and noticed that it does not pass a model_path into the Tracker() constructor. Looking at past commits, I see you made a change to predict.py to default the model_path argument to none. I think that change may be causing this error when running predict_ros.py...
` /home/se3_tracknet/pose_init.txt
Traceback (most recent call last):
  File "predict_ros.py", line 109, in <module>
    tracker = Tracker(dataset_info, images_mean, images_std,ckpt_dir,trans_normalizer=dataset_info['max_translation'],rot_normalizer=dataset_info['max_rotation'])
  File "/home/se3_tracknet/predict.py", line 131, in __init__
    mesh = trimesh.load(model_path)
  File "/usr/local/lib/python3.6/dist-packages/trimesh/exchange/load.py", line 159, in load
    file_type)
ValueError: File type: nonetype not supported
`

wenbowen123, 2024-01-23:
Thanks for bringing this up! Like you said, it is possible that the change for the `predict.py` did not take into account the `predict_ros.py`. First of all, did you successfully run the `predict.py` for the public datasets?

wenbowen123, 2024-01-25:
Great to hear that you got success running the examples. For ros, you might need a bit modification, such as providing `model_path` which is the path to the CAD model (refer to the example)