wenbowen123/iros20-6d-pose-tracking #73: Blender script not working with newer Blender versions

aarist, 2024-01-26:
Hello,
Could you please provide details about the Blender and its Python version used for the data generation?
The Blender API was massively updated since Blender 2.8, so the `blender_dataset_generator.py ` script doesn't run on newer versions, such as 3.3. I tried Blender 2.79, which comes with Python 3.5. However, I get confusing errors with opencv-python and numpy version mismatch.
Best,
Andi

wenbowen123, 2024-01-26:
Hi, we will soon release [FoundationPose](https://nvlabs.github.io/FoundationPose/) that does not need your own training. It can be generalized to your own object.

aarist, 2024-01-26:
Thank you for your fast response!
I am looking forward to work with FoundationPose, however I would be really thankful if you could still provide the requested versions. I already tested the models you trained on YCB-objects in my lab with our YCB-objects, and it works like a charm in real-time for my application. For custom objects, I need the synthetic data generation of se(3)-Tracknet, which is currently not possible without knowing the exact versions : -)

wenbowen123, 2024-01-26:
great that it works well for you! We used blender 2.79. It's a bit out-dated, so you need to find a matched version for opencv. Also to note blender uses its own python so it might sometimes confuse with the python on your system.

aarist, 2024-02-05:
Thank you! The versions that worked for me with Blender 2.79 were opencv-python==3.4.2.17 and numpy==1.14.5 : -)

closed 2024-02-05 by wenbowen123