wenbowen123/iros20-6d-pose-tracking #74: building the dockerfile didn't work

monajalal, 2024-01-29:
Could you please confirm if the dockerfile may work for building in Ubuntu 22.04?

```
(base) mona@ada:~/iros20-6d-pose-tracking/docker$ docker build -t posetrack .
[+] Building 205.4s (7/10)                                                                                                                                                                   docker:default
 => [internal] load build definition from dockerfile                                                                                                                                                   0.0s
 => => transferring dockerfile: 2.76kB                                                                                                                                                                 0.0s
 => [internal] load .dockerignore                                                                                                                                                                      0.0s
 => => transferring context: 2B                                                                                                                                                                        0.0s
 => [internal] load metadata for docker.io/nvidia/cudagl:11.3.0-devel-ubuntu16.04                                                                                                                      1.0s
 => [1/7] FROM docker.io/nvidia/cudagl:11.3.0-devel-ubuntu16.04@sha256:01ed8befe59f7d9c967ff84a4ced41af5dfecf5028d415579e5837a9ce7425b6                                                               43.8s
 => => resolve docker.io/nvidia/cudagl:11.3.0-devel-ubuntu16.04@sha256:01ed8befe59f7d9c967ff84a4ced41af5dfecf5028d415579e5837a9ce7425b6                                                                0.0s
 => => sha256:01ed8befe59f7d9c967ff84a4ced41af5dfecf5028d415579e5837a9ce7425b6 4.71kB / 4.71kB                                                                                                         0.0s
 => => sha256:fb15d46c38dcd1ea0b1990006c3366ecd10c79d374f341687eb2cb23a2c8672e 170B / 170B                                                                                                             0.2s
 => => sha256:99af895351a7df2761a976118ad5b6600bf64ebdd5190c87e36236783bd273cf 20.34kB / 20.34kB                                                                                                       0.0s
 => => sha256:b51569e7c50720acf6860327847fe342a1afbe148d24c529fb81df105e3eed01 857B / 857B                                                                                                             0.1s
 => => sha256:58690f9b18fca6469a14da4e212c96849469f9b1be6661d2342a4bf01774aa50 46.50MB / 46.50MB                                                                                                       0.8s
 => => sha256:da8ef40b9ecabc2679fe2419957220c0272a965c5cf7e0269fa1aeeb8c56f2e1 528B / 528B                                                                                                             0.1s
 => => sha256:df762380fec23daccc056c85c688be0a40f08ebca1c0f4bcfe593870c94221eb 6.84MB / 6.84MB                                                                                                         0.9s
 => => sha256:35e2da6314f0f15f4c56cbcc088671ac47fdadbfa49e0cffc7d4af4115bf8252 11.30MB / 11.30MB                                                                                                       1.5s
 => => sha256:b1876ed457134301a7df34a3ca2439dcfc33697041b82fb1f807a6581848f844 187B / 187B                                                                                                             1.1s
 => => extracting sha256:58690f9b18fca6469a14da4e212c96849469f9b1be6661d2342a4bf01774aa50                                                                                                              2.0s
 => => sha256:ef2c706bd6871a3583243de994a8002f078d188a65da183d78eca217488d1662 6.43kB / 6.43kB                                                                                                         1.2s
 => => sha256:8fdfcb70e9325ac4ed818c1ea97b53c196269a921715244a355ca8c4ab9c31ad 867.59MB / 867.59MB                                                                                                    21.2s
 => => sha256:fd49abd7a85d02440054713ad46677f045be713f83dd5df7ffe621125cd26b07 72.12kB / 72.12kB                                                                                                       1.5s
 => => sha256:5baeeb1047941986575874fbb938a4d489df9a9c4ca28080d032ad6fd1d13184 1.06GB / 1.06GB                                                                                                        24.2s
 => => sha256:f19e8bdcbaf67dc874fb1695ffcb6051b6593a61b5f8509797c7e9f20c729e76 95.54kB / 95.54kB                                                                                                       1.7s
 => => sha256:486d17129b8941ec0929501ef484ba2c6396a34145922c690014dace453cb0c7 6.52MB / 6.52MB                                                                                                         2.7s
 => => sha256:48f76cdc2d1191d0d835d8a868afc8ccf6c92940f5ddb7006af501ae79bdc91b 191B / 191B                                                                                                             3.2s
 => => extracting sha256:b51569e7c50720acf6860327847fe342a1afbe148d24c529fb81df105e3eed01                                                                                                              0.0s
 => => extracting sha256:da8ef40b9ecabc2679fe2419957220c0272a965c5cf7e0269fa1aeeb8c56f2e1                                                                                                              0.0s
 => => extracting sha256:fb15d46c38dcd1ea0b1990006c3366ecd10c79d374f341687eb2cb23a2c8672e                                                                                                              0.0s
 => => extracting sha256:df762380fec23daccc056c85c688be0a40f08ebca1c0f4bcfe593870c94221eb                                                                                                              0.3s
 => => sha256:8c6ebfbd95c0beb9bcff475311ee30886b20b7cecf3d50354aa66e2179ee63e0 358.01kB / 358.01kB                                                                                                     3.5s
 => => extracting sha256:35e2da6314f0f15f4c56cbcc088671ac47fdadbfa49e0cffc7d4af4115bf8252                                                                                                              0.3s
 => => extracting sha256:b1876ed457134301a7df34a3ca2439dcfc33697041b82fb1f807a6581848f844                                                                                                              0.0s
 => => extracting sha256:ef2c706bd6871a3583243de994a8002f078d188a65da183d78eca217488d1662                                                                                                              0.0s
 => => sha256:67b698bad598328472c437453034c5b8a2370fd61da5133929aab1fd0454d451 325.73kB / 325.73kB                                                                                                     4.2s
 => => sha256:8136db01146f279fa8312662fee58233c54ad88e99262ecf2c09d00568b0905c 316B / 316B                                                                                                             4.4s
 => => sha256:0c2b8fa50c12cd08a4783986cf7e0cb2ee765f723437c45c2ff740147cece2c4 8.37kB / 8.37kB                                                                                                         4.7s
 => => sha256:4e3658e629d04b07b86ea085b6acbd284ffc436f54f074e247e92f8157c13a26 4.90MB / 4.90MB                                                                                                         5.6s
 => => sha256:bcd6549196e65de4e2cd4668cd3299b4f54d91a2cc0252b5533f75c13600580a 239.58kB / 239.58kB                                                                                                     5.9s
 => => sha256:a2644530c9f7eead0a04603678bb3e67384bf3ebaecc89e5140514afd8e34993 522B / 522B                                                                                                             6.2s
 => => extracting sha256:8fdfcb70e9325ac4ed818c1ea97b53c196269a921715244a355ca8c4ab9c31ad                                                                                                              9.5s
[... cut ...]
  26 | >>>   python3 -m pip install Pillow --upgrade
  27 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c rm -rf /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so &&  python3 -m pip install --upgrade pip &&  python3 -m pip install trimesh==3.7.12 open3d==0.9.0.0 opencv-python  transformations torchviz torchsummary vispy==0.6.4 PyOpenGL==3.1.0 plyfile pyglet==1.2.4 pyrender==0.1.43 scikit-image==0.16.2 &&  python3 -m pip install Pillow --upgrade" did not complete successfully: exit code: 1

```

aThinkingNeal, 2024-01-31:
Greetings, I am facing the issue while using the dockerfile to build the image

[image]

wenbowen123, 2024-02-01:
It should has nothing to do with Ubuntu 22. What if you just do `pip install pyrender` and skip `pyglet`

ZisongXu, 2024-02-19:
Hi Dear bowen, I tried just do `pip install pyrender` and skip `pyglet`, it seems work, and now i am trying continue to do some test. If i have any questions i will give you feed back

monajalal, 2024-02-19:
@wenbowen123 thanks a lot for your response. Confirming that it also worked for me.

```
(base) mona@ada:~/iros20-6d-pose-tracking/docker$  docker build -t posetrack .
[+] Building 1432.2s (11/11) FINISHED                                                                                                                                                        docker:default
 => [internal] load build definition from dockerfile                                                                                                                                                   0.0s
 => => transferring dockerfile: 2.75kB                                                                                                                                                                 0.0s
 => [internal] load .dockerignore                                                                                                                                                                      0.0s
 => => transferring context: 2B                                                                                                                                                                        0.0s
 => [internal] load metadata for docker.io/nvidia/cudagl:11.3.0-devel-ubuntu16.04                                                                                                                      5.6s
 => [1/7] FROM docker.io/nvidia/cudagl:11.3.0-devel-ubuntu16.04@sha256:01ed8befe59f7d9c967ff84a4ced41af5dfecf5028d415579e5837a9ce7425b6                                                                0.0s
 => CACHED [2/7] RUN apt-get update &&  apt-get install -y software-properties-common &&  add-apt-repository ppa:deadsnakes/ppa &&  add-apt-repository ppa:ubuntu-toolchain-r/test &&  add-apt-reposi  0.0s
 => CACHED [3/7] RUN apt-get install -y python3.6 python3.6-dev &&  ln -sf /usr/bin/python3.6 /usr/local/bin/python3 &&  ln -sf /usr/bin/python3.6 /usr/bin/python3 &&  cd / && wget https://bootstra  0.0s
 => [4/7] RUN  rm -rf /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so &&  python3 -m pip install --upgrade pip &&  python3 -m pip install trimesh==3.7.12 open3d==0.9.0.0 opencv-python  trans  1300.6s
 => [5/7] RUN python3 -m pip install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html                               70.1s 
 => [6/7] RUN cd / && wget https://download.blender.org/release/Blender2.79/blender-2.79b-linux-glibc219-x86_64.tar.bz2  &&    tar xvf blender-2.79b-linux-glibc219-x86_64.tar.bz2 && rm -rf blender  14.7s 
 => [7/7] RUN cd / && rm -rf get-pip.py && wget https://bootstrap.pypa.io/pip/3.5/get-pip.py &&    /blender-2.79b-linux-glibc219-x86_64/2.79/python/bin/python3.5m get-pip.py &&    rm -rf /blender-  18.3s 
 => exporting to image                                                                                                                                                                                22.9s 
 => => exporting layers                                                                                                                                                                               22.9s 
 => => writing image sha256:a91070b8c9e2dd280bfa98abb53a87712e6665442f2e5ff7226a5cb85d866a6d                                                                                                           0.0s 
 => => naming to docker.io/library/posetrack  
```

closed 2024-02-19 by monajalal