wenbowen123/iros20-6d-pose-tracking #76: RuntimeError: Attempting to deserialize object on a CUDA device but torch.cuda.is_available() is False. If you are running on a CPU-only machine, please use torch.load with map_location=torch.device('cpu') to map your storages to the CPU.

ZisongXu, 2024-02-20:
Hi Dear Bowen:

I am coming from your another repository "BundleSDF", and you recommend me to use "se3-tracknet".

I am trying re-run your code but I got this issue:
`
RuntimeError: Attempting to deserialize object on a CUDA device but torch.cuda.is_available() is False. If you are running on a CPU-only machine, please use torch.load with map_location=torch.device('cpu') to map your storages to the CPU.`

Because our lab uses the singularity container, I would like to reproduce your project in singularity first. When I built the singularity container, I changed the image to the docker image you provided and it seemed that the build was successful. But when I run your example code it had the above problem.

I am thinking that my GPU is 4090, could it be that the version of cuda and cudnn is too low, causing the GPU to be unable to be used? Because after I created the container, I found that it was ubuntu16 and the cuda version was 10.1.

If your code supports GPU4090, maybe I should use docker containers. But I just wanted to try singularity first.

Best Regards
Zisong Xu

ZisongXu, 2024-02-20:
@wenbowen123 Ok thanks a lot!!! Can I ask when will you upload the code of [FoundationPose](https://github.com/NVlabs/FoundationPose), I realized that I can see the paper on arXive.

ZisongXu, 2024-02-20:
@wenbowen123  Oh sorry, by the way do I also need to change the version of Ubuntu from 16 to 20?

wenbowen123, 2024-02-20:
Our target release is in March/April.

Re Ubuntu version, I haven't tried that. If you run into dependency issues that expect certain OS version, might worth changing that too.

ZisongXu, 2024-02-21:
Thanks a lot!

closed 2024-02-21 by ZisongXu

wenbowen123, 2024-03-28:
@ZisongXu Our FoundationPose code has released.