wenbowen123/iros20-6d-pose-tracking #77: TypeError: Unexpected keyword argument {'n_jobs': 10}

ZisongXu, 2024-03-03:
Hi Dear Bowen:

I think I let your se3tracknet project work in the singularity container with GPU 3080.

[image]

But at last I got some errors:
```

>>>>>>>>>>>>>>>> 1300
>>>>>>>>>>>>>>>> 1400
>>>>>>>>>>>>>>>> 1500
>>>>>>>>>>>>>>>> 1600
>>>>>>>>>>>>>>>> 1700
>>>>>>>>>>>>>>>> 1800
>>>>>>>>>>>>>>>> 1900
>>>>>>>>>>>>>>>> 2000
Traceback (most recent call last):
  File "predict.py", line 668, in <module>
    predictSequenceYcb()
  File "predict.py", line 578, in predictSequenceYcb
    err = U.adi(pred_poses[i],gt_poses[i],tracker.object_cloud)
  File "/home/zisongxu/iros20-6d-pose-tracking/Utils.py", line 96, in adi
    nn_dists, _ = nn_index.query(np.asarray(gt_model.points).copy(), k=1, n_jobs=10)
  File "_ckdtree.pyx", line 783, in scipy.spatial._ckdtree.cKDTree.query
  File "_ckdtree.pyx", line 387, in scipy.spatial._ckdtree.get_num_workers
TypeError: Unexpected keyword argument {'n_jobs': 10}
```
Is that right or how can I solve it.

Best Regards
Zisong Xu

ZisongXu, 2024-03-04:
Oh thanks I think it works right?

```
>>>>>>>>>>>>>>>> 1300
>>>>>>>>>>>>>>>> 1400
>>>>>>>>>>>>>>>> 1500
>>>>>>>>>>>>>>>> 1600
>>>>>>>>>>>>>>>> 1700
>>>>>>>>>>>>>>>> 1800
>>>>>>>>>>>>>>>> 1900
>>>>>>>>>>>>>>>> 2000
reinit_frames , adi_auc 97.8769132312819
[se3track] Singularity> ~/iros20-6d-pose-tracking $ 
```

Best Regards
Zisong Xu

wenbowen123, 2024-03-04:
Nice

closed 2024-03-04 by wenbowen123