wenbowen123/iros20-6d-pose-tracking #78: Problem with Generate my own data

ZisongXu, 2024-03-04:
Hi Dear Bowen:

Sorry to bother you again, I am trying to generate my own data, but I met some problems below:
```

[se3track] Singularity> ~/iros20-6d-pose-tracking $ python3 blender_main.py
================
/blender-3.3.0-linux-x64/blender /home/zisongxu/iros20-6d-pose-tracking/object_models/bunny/1.blend -b -P /home/zisongxu/iros20-6d-pose-tracking/blender_dataset_generator.py 
================
Blender 3.3.0 (hash 0759f671ce1f built 2022-09-07 00:16:43)
Failed to create secure directory (/tmp/pulse): No such file or directory
WARN (bke.appdir): source/blender/blenkernel/intern/appdir.c:1126 tempdir_session_create: Could not generate a temp file name for '/tmp/blender_dOpwPe', falling back to '/tmp/'
/tmp/gvfs/ non-existent directory
Read blend: /home/zisongxu/iros20-6d-pose-tracking/object_models/bunny/1.blend
Color management: scene view "sRGB OETF" not found, setting default "Standard".
Color management: sequencer colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Color management: image colorspace "sRGB OETF" not found, will use default instead.
Error: Python: SyntaxError: encoding problem: future_fstrings
rmdir: Read-only file system

Blender quit
0
[se3track] Singularity> ~/iros20-6d-pose-tracking $ 
```

I followed your instruction to modify "dataset_info.yml", should I also change other files? Like:

```
Failed to create secure directory (/tmp/pulse): No such file or directory
WARN (bke.appdir): source/blender/blenkernel/intern/appdir.c:1126 tempdir_session_create: Could not generate a temp file name for '/tmp/blender_dOpwPe', falling back to '/tmp/'
/tmp/gvfs/ non-existent directory
```
I do not know where to find them or edit them?

Thanks a lot!!

Best Regards
Zisong Xu

wenbowen123, 2024-03-06:
do you have `/tmp` dir on your system? Do you have permission to create folders there?

ZisongXu, 2024-03-11:
I am so sorry Bowen, I am a little busy recently, I will try to solve this problem soon!

Best Regards
Zisong XU