wenbowen123/iros20-6d-pose-tracking #81: SET UP PROBLEM

Hitlzw, 2024-03-12:
[image]
what is the problem?Should I install nvidia-docker?

wenbowen123, 2024-03-13:
yes, have you tried that?

Hitlzw, 2024-03-14:
> yes, have you tried that?

Thank you, professor, at the moment I have been able to run the code locally without any problems. But I wonder how much GPU memory the model will use.Only 246MiB?
[image]

wenbowen123, 2024-03-14:
I dont remember the exact memory usage. But it is indeed a light-weight network and consumes less memory. Closing since the issue has been resolved.

closed 2024-03-14 by wenbowen123