wenbowen123/iros20-6d-pose-tracking #83: generate data problem

Hitlzw, 2024-03-19:
When I run the python blender_main.py command, I can generate data normally when I don't modify dataset_info.yml. But when I changed focalX, focalY, centerX, centerY in dataset_info.yml to the parameters of my real-world camera, the resulting data got weird. The resolution of a real-world camera is also 640x480.
[image]
[image]
[image]

wenbowen123, 2024-03-19:
Hi, it has been a long time since I worked on this part. I'd encourage to checkout our recent work [FoundationPose](https://github.com/NVlabs/FoundationPose) which can be immediately applied to your setup without retraining.

Hitlzw, 2024-03-20:
Thank you for your reply. I think the cause of the problem is that the newly set camera parameters caused the camera FOV to become larger, and the camera FOV exceeded the scope of the box in 1.blender file (I don't know much about blender operation), which caused the rgb graphics and seg images to become abnormal. And I've been watching FoundationPose for a while, looking forward to its code.
[image]

Hitlzw, 2024-03-20:
> Hi, it has been a long time since I worked on this part. I'd encourage to checkout our recent work [FoundationPose](https://github.com/NVlabs/FoundationPose) which can be immediately applied to your setup without retraining.

Thank you for your reply.  I think the cause of the problem is that the newly set camera parameters caused the camera FOV to become larger, and the camera FOV exceeded the scope of the box in 1. blender file (I don't know much about blender operation), which caused the rgb graphics and seg images to become abnormal.  And I've been watching FoundationPose for a while, looking forward to its code.
[image]

closed 2024-03-21 by Hitlzw